VSChina/ESP32_AzureIoT_Arduino #5: Error initializing SNTP

jorgevelascoperez, 2018-06-20:
Hi,

An error occurs in the two examples if the delay between messages of 5 minutes or more.

"start sending events.
Info: >>>IoTHubClient_LL_SendEventAsync accepted message for transmission to IoT Hub.

Error: Time:Wed Jun 20 12:41:38 2018 File:C:\Users\jorge\Documents\Arduino\libraries\azure_esp32\src\Esp32MQTTClient.cpp Func:SendEventOnce Line:316 Waiting for send confirmation, time is up 10003

Info: >>>Re-connect.

Info: >>>Confirmation[1] received for message tracking id = 1 with result = IOTHUB_CLIENT_CONFIRMATION_BECAUSE_DESTROY

Info: Initializing SNTP

assertion "Operating mode must not be set while SNTP client is running" failed: file "/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/apps/sntp/sntp.c", line 591, function: sntp_setoperatingmode
abort() was called at PC 0x400e01eb on core 1

Backtrace: 0x40087b34:0x3ffd53d0 0x40087c33:0x3ffd53f0 0x400e01eb:0x3ffd5410 0x400f23d9:0x3ffd5440 0x400d3a3e:0x3ffd5460 0x400d3a9a:0x3ffd5480 0x400d2b96:0x3ffd54a0 0x400d2e3e:0x3ffd54e0 0x400d2e75:0x3ffd5500 0x400d2f9b:0x3ffd5540 0x400d22c1:0x3ffd5560 0x4012da3c:0x3ffd5600

Rebooting..."

pgrlopes, 2018-07-03:
Hello sir, did you manage to solve your issue? I'm having the exact same thing happen to me..

jorgevelascoperez, 2018-07-04:
I have not been able to solve it. I was waiting for some update, but
it seems that these libraries have no maintenance. The problem is that
they are the only libraries available, I hope to find a solution soon.

A greeting

pgrlopes, 2018-07-05:
Well someone using the ESP IDF with the same problem managed to solve it, but nothing with the arduino ide and yeah, these libraries seem to be abandoned.
Welp, onto the next cloud service I guess..

pgrlopes, 2018-07-18:
Yeah library is dead, you need to try and use the ESP-IDF I believe

lirenhe, 2018-07-20:
The  Azure IoT library for ESP32 is coming from https://github.com/espressif/esp-azure.
We created this library for testing purpose for people who want  to try Azure IoT in Arduino for ESP32 boards .
We put the source code in GitHub as we hope to leverage community effort to improve the quality of it.

For the problem you face, the connection between device and Iothub could be closed if there is no active connection for sometime, so the device is unable to send messages.

Could you try to add the follow code in int SNTP_Init()  in \arduino-esp32\libraries\AzureIoT\src\az_iot\c-utility\pal\lwip\sntp_lwip.c

    if(sntp_enabled()){
        sntp_stop();
    }

samisarfraz, 2018-07-20:
thx lirenhe, i have tried your suggestion and added it to library and kept it running and after about 40 minutes i am getting an error on serial monitor:

Info: Initializing SNTP

assertion "Operating mode must not be set while SNTP client is running" failed: file "/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/apps/sntp/sntp.c", line 591, function: sntp_setoperatingmode
abort() was called at PC 0x400e307f on core 1

Backtrace: 0x4008dd1c:0x3ffd1000 0x4008def7:0x3ffd1020 0x400e307f:0x3ffd1040 0x40113b45:0x3ffd1070 0x400d4146:0x3ffd1090 0x400d41a2:0x3ffd10b0 0x400d318a:0x3ffd10d0 0x400d34be:0x3ffd1110 0x400d3625:0x3ffd1130 0x400d26f7:0x3ffd1150 0x4013d296:0x3ffd1290

Rebooting...

samisarfraz, 2018-07-21:
Hello ive tried changed the position of lines in the files you mentioned above and now library is working perfectly fine 

in the file \arduino-esp32\libraries\AzureIoT\src\az_iot\c-utility\pal\lwip\sntp_lwip.c
add these lines

if(sntp_enabled()){
    sntp_stop();
}

before return 0; 

Thanks for the help lirenhe.

anthrene, 2018-07-21:
I think that to edit sntp_lwip.c is not good manner and this issue should be fixed in source for ESP32, not in c-utility source.

In Esp32MQTTClient.cpp, platform_init() is called in Esp32MQTTClient_Init func. So I think that it is better way to call platform_deinit() in Esp32MQTTClient_Close func like below.

`void Esp32MQTTClient_Close(void)
{
    if (iotHubClientHandle != NULL)
    {
        IoTHubClient_LL_Destroy(iotHubClientHandle);
        iotHubClientHandle = NULL;
    }

    platform_deinit();
}` 

As long as Esp32MQTTClient_Init func and Esp32MQTTClient_Close func are called proper manner, this edit works.
And ofcourse works when CheckConnection func is called for reset use.

>lirenhe

I made the pull request to fix this issue by the edit. Please check it.

PR #9 (2018-07-31): Bug fix in reset procedure.
This is the pull request to fix #5 issue.

I Insert platform_deinit() call in Esp32MQTTClient_Close func.
As long as Esp32MQTTClient_Init func and Esp32MQTTClient_Close func are called proper manner, this edit works.
And ofcourse works when CheckConnection func is called for reset use.
closed 2018-07-31 by lirenhe