VSChina/ESP32_AzureIoT_Arduino #7: "Operating mode must not be set while SNTP client is running" on sntp.c

pgrlopes, 2018-07-03:
Hello.

I tried using both getstarted and MQTT examples but I keep getting this error when trying to access the Hub. My connection string is as follows:

static const char* connectionString = "HostName=<My iot hub name>;DeviceId=<My device name>;SharedAccessSignature=<My key";

I checked on a thread with the same issue where increasing the time in "platform_init" function solved the issue but I can't find that function in this implementation.

Any help would be appreciated.

Cheers

lirenhe, 2018-07-20:
platform_init is defined here:  https://github.com/VSChina/ESP32_AzureIoT_Arduino/blob/master/src/az_iot/c-utility/pal/src/platform_openssl_compact.c

pgrlopes, 2018-08-01:
The latest changes solved the issue, thank you.

closed 2018-08-14 by pgrlopes