VSChina/ESP32_AzureIoT_Arduino #8: Cannot connect to Iot Hub

KoRba88, 2018-07-10:
Hi 

I have compiled and uploaded "GetStarted" example without errors BUT

Serial monitor shows these messeges 

Error: Time:Tue Jul 10 19:48:41 2018 File:C:\Users\Marcin\Documents\Arduino\libraries\ESP32_AzureIoT_Arduino\src\az_iot\c-utility\src\base64.c Func:Base64_Decoder Line:201 Invalid length Base64 string!

Error: Time:Tue Jul 10 19:48:41 2018 File:C:\Users\Marcin\Documents\Arduino\libraries\ESP32_AzureIoT_Arduino\src\az_iot\c-utility\src\sastoken.c Func:construct_sas_token Line:217 Unable to decode the key for generating the SAS.

Error: Time:Tue Jul 10 19:48:41 2018 File:C:\Users\Marcin\Documents\Arduino\libraries\ESP32_AzureIoT_Arduino\src\az_iot\iothub_client\src\iothub_client_authorization.c Func:IoTHubClient_Auth_Get_SasToken Line:389 Failed creating sas_token

Error: Time:Tue Jul 10 19:48:41 2018 File:C:\Users\Marcin\Documents\Arduino\libraries\ESP32_AzureIoT_Arduino\src\az_iot\iothub_client\src\iothubtransport_mqtt_common.c Func:SendMqttConnectMsg Line:1585 failure getting sas token from IoTHubClient_Auth_Get_SasToken.

Info: >>>Re-connect.

Info: Initializing SNTP

assertion "Operating mode must not be set while SNTP client is running" failed: file "/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/apps/sntp/sntp.c", line 591, function: sntp_setoperatingmode
abort() was called at PC 0x400e2cf7 on core 1

Backtrace: 0x40088348:0x3ffd0bd0 0x40088447:0x3ffd0bf0 0x400e2cf7:0x3ffd0c10 0x401135bd:0x3ffd0c40 0x400d3f5e:0x3ffd0c60 0x400d3fba:0x3ffd0c80 0x400d2fa2:0x3ffd0ca0 0x400d32d6:0x3ffd0ce0 0x400d343d:0x3ffd0d00 0x400d2653:0x3ffd0d20 0x401384b4:0x3ffd0e60

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13076
entry 0x40078a58
ESP32 Device
Initializing...
 > WiFi
Connecting...
....WiFi connected
IP address: 
192.168.1.74
 > IoT Hub
Info: Initializing SNTP

Info: SNTP initialization complete

Info: IoT Hub SDK for C, version 1.1.23

samisarfraz, 2018-07-17:
Its is keep on reconnecting and rebooting ESP 32 device. Having similar issue.

lirenhe, 2018-07-20:
From the error, it seems you did not provide the correct device connection string.   It failed with the following check 

        if ((strlen(source) % 4) != 0)
        {
            /*Codes_SRS_BASE64_06_011: [If the source string has an invalid length for a base 64 encoded string then Base64_Decode shall return NULL.]*/
            LogError("Invalid length Base64 string!");
            result = NULL;
        }

Could you double check the device key in the connection string?

pgrlopes, 2018-08-01:
Hello @lirenhe 

I have a similar issue. I created a device within my IoT hub in the azure portal and I'm providing the signature string as it is given to me in the device panel, so something like this:

"HostName=<my_iothub_name.azure-devices.net>;DeviceId=<MyDeviceID>;SharedAccessKey=<Device_PrimaryKey>"

It does not work though, saying that my Base64 code has invalid length. By checking a couple of files I can see that the SAS token is retrieved from the key, so I tried changing the SharedAccessKey in the string to SharedAccessSignature. That also does not work, it just says ">>> Connection Status: timeout" and this message is given because the authentication failed in the ConnectionStatusCallback function from the ESP32MQTTClient.cpp file.

Much like OP, im probably doing something trivial very wrong, would appreciate a bit of help understanding the authentication process.

Thank you for your time.

KoRba88, 2018-08-01:
I solved this issue.  "HostName=<my_iothub_name.azure-devices.net>;DeviceId=;SharedAccessKey=<Device_PrimaryKey>" should be written without <>. Just directly copy it from azure iot hub account.

lirenhe, 2018-08-02:
@pgrlopes , you need to remove '<>' marks in your connection string.   Do check this doc: https://blogs.msdn.microsoft.com/iotdev/2017/05/09/understand-different-connection-strings-in-azure-iot-hub/