VSChina/ESP32_AzureIoT_Arduino #24: Get started not working with esp32

ashwanisihag, 2021-04-27:
IP address: 
192.168.41.128
[  4634][E][esp32-hal-adc.c:164] __analogRead(): GPIO0: ESP_ERR_TIMEOUT: ADC2 is in use by Wi-Fi.
Info: Initializing SNTP

Info: SNTP initialization complete

Info: IoT Hub SDK for C, version 1.1.23

Error: Time:Tue Apr 27 08:58:50 2021 File:C:\Users\Vaeg\Documents\Arduino\libraries\ESP32_AzureIoT_Arduino\src\az_iot\c-utility\pal\socket_async.c Func:socket_async_create Line:134 Socket connect failed, not EINPROGRESS: 0

Info: Could not open the socket

Info: >>>Connection status: disconnected

Info: >>>Connection status: disconnected

Info: tlsio_openssl_close has been called when in neither TLSIO_STATE_OPEN nor TLSIO_STATE_ERROR.

cyberwaelder1, 2021-08-25:
hello, is there a solution? I have the same problem.

ftylitak, 2022-04-04:
Even though this is quite old, just for future reference, this issue is caused because you try to use this library with esp-idf 4.x which is not supported based on my tests (or ESP32 Arduino 2.x). 

Switch back to esp-idf 3.x (3.3.6 is the newest at the time of writing).

If you are using Arduino code and Arduino IDE, you need to go to the Boards Manager and downgrade esp32 support to maximum version 1.0.6.

ttrichar, 2022-09-20:
> A very rough workaround is to comment out all the following lines:
> 
> https://github.com/VSChina/ESP32_AzureIoT_Arduino/blob/0c090d0832b165078fbe877012e4cf62f42aeb24/src/az_iot/c-utility/pal/socket_async.c#L128-L152
> 
> and replace them by:
> 
> ```c++
> result = sock;
> ```
> 
> May be it is a timing issue. Sounds bad not to check for the socket connect return code...though it works for me with esp-idf v4.4.

This worked for me, thank you!